ControlCore-Project/concore #23: Issue in installation command

shivangvijay, 2022-12-24:
I follow the docs and use the command "pip install -r requirements.txt", it gives the error:- 

ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-5JycN5/cvxopt/setup.py'"'"'; __file__='"'"'/tmp/pip-install-5JycN5/cvxopt/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-6fFlfg
         cwd: /tmp/pip-install-5JycN5/cvxopt/
    Complete output (8 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-5JycN5/cvxopt/setup.py", line 4, in <module>
        import versioneer
      File "versioneer.py", line 369
        LONG_VERSION_PY: Dict[str, str] = {}
                       ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

shivangvijay, 2022-12-24:
"pip3 install -r requirements.txt" works.

pradeeban, 2023-01-06:
Good to know you have got this working.

Yes, you may need to replace "pip" with "pip3" and "python" with "python3" in the documentation based on your installation.

closed 2023-01-06 by pradeeban