ControlCore-Project/concore #25: ../editgraph is not working

shivangvijay, 2023-01-07:
Attaching the screenshot 
[image]

amit-62, 2023-01-11:
`editgraph` opens controlcore-editor you can alternatively open it using this [link](http://controlcore-project.github.io/DHGWorkflow).
Try `getgraph` see if it work, don't forget to follow instruction guide.

pradeeban, 2023-01-11:
Thanks @shivangvijay for pointing out this bug and the fix.

This is fixed in main and dev, crediting to you in the commit - https://github.com/ControlCore-Project/concore/commit/4e708b0a6ee6255ea7fc0b8d13568ffd802aa648

Please let us know if this problem still persists.

shivangvijay, 2023-01-12:
Thanks, It's working now.

closed 2023-01-12 by shivangvijay

pradeeban, 2023-01-12:
Thanks a lot, @shivangvijay, for all your contributions!