ControlCore-Project/concore #27: https://controlcore-project.github.io/DHGWorkflow/ is not working

shivangvijay, 2023-01-07:
In dev branch:- https://controlcore-project.github.io/DHGWorkflow/ link is used in editgraph, which is not working.

pradeeban, 2023-01-08:
Thanks for the report. The links should be https://controlcore-project.github.io/concore-editor/

Let me fix it.

shivangvijay, 2023-01-08:
Please check my pull request #26. I corrected link and changes the code to check OS (Linux/MAC).

pradeeban, 2023-01-08:
Thanks for the attempt at fixing this. I have closed the pull request as it has some issues. I have fixed the issue by replacing the file with the one in the main branch (which is the later branch).

Also, please work on the main branch, as the dev branch appears stale now. If we need you to work on the dev branch, we will suggest you so explicitly.

The "dev" branch is more like an experimental branch than a dev branch, for concore.

closed 2023-01-08 by pradeeban

pradeeban, 2023-01-08:
I have fixed the README, as it was suggesting to use the dev branch. It is not needed. All the important changes from the dev are already in the main branch, and the dev branch appears stale at this point.