ControlCore-Project/concore #31: build route not working (doubt)

parteekcoder, 2023-02-18:
I am trying to make request on all routes to understand the concore better "upload" route is working fine 
But when I ran "build" it didn't work 

It showed me "FileNotFoundError" see last line

[image]

@pradeeban

parteekcoder, 2023-02-18:
Hey @pradeeban This is a bug.

I am running this on windows

I will create a PR for this

pradeeban, 2023-02-18:
Can you please re-run on the main branch and copy-paste the whole error (instead of a screenshot)?

@amit-62 implemented this and tested it on Linux and Mac. It was not tested on Windows, and likely, something was missing. Any possible fixes should be tested on both Linux and Windows to avoid breaking the Linux version.

parteekcoder, 2023-02-18:
> Can you please re-run on the main branch and copy-paste the whole error (instead of a screenshot)?
> 
> @amit-62 implemented this and tested it on Linux and Mac. It was not tested on Windows, and likely, something is missing. Any possible fixes should be tested on both Linux and Windows to avoid breaking the Linux version.

I have tested my changes on windows and MACOS but I will update all paths shortly

pradeeban, 2023-02-18:
Thanks a lot, @parteekcoder!

parteekcoder, 2023-02-18:
> Can you please re-run on the main branch and copy-paste the whole error (instead of a screenshot)?
>
> @amit-62 implemented this and tested it on Linux and Mac. It was not tested on Windows, and [...]

yes sure this is the error

Traceback (most recent call last):
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\site-packages\flask\app.py", line 2070, in wsgi_app
    response = self.full_dispatch_request()
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\site-packages\flask\app.py in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified127.0.0.1 - - [19/Feb/2023 01:34:44] "POST /build/test?fetch=sampleplot&apikey=xyz HTTP/1.1" 500 -
[2023-02-19 01:34:59,867] ERROR in app: Exception on /build/test [POST]
Traceback (most recent call last):
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\site-packages\flask\app.py", line 2070, in wsgi_app
    response = self.full_dispatch_request()
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\site-packages\flask\app.py", line 1515, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\site-packages\flask_cors\extension.py", line 165, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\site-packages\flask\app.py", line 1513, in full_dispatch_request
    rv = self.dispatch_request()
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\site-packages\flask\app.py", line 1499, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "F:\GSOC\emori-bmi-gsoc\concore\concore\fri\server\main.py", line 73, in build
    proc = call(["./makestudy", makestudy_dir], cwd=concore_path)
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 345, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Nitika\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 1438, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

pradeeban, 2023-02-22:
Fixed in #51.

closed 2023-02-22 by pradeeban

amit-62, 2023-02-22:
Hello everyone sorry for chiming in late but putting shell=true exposes it security leaks. @Rahuljagwani please look into [this](https://stackoverflow.com/questions/3172470/actual-meaning-of-shell-true-in-subprocess).