ControlCore-Project/concore #34: Clear is running fine But Actually not clearing the files (bug MAC OS)

parteekcoder, 2023-02-18:
Reproduce 
1) build and debug a study
2) try to clear it using clear route

Expected 
should clean the results

Actual 

Request giving good response but actually not deleting files

parteekcoder, 2023-02-18:
Actually the Problem is that When bat files are converting into scripts in MACOS

rm CU/*
And when it is running this, It shows no File or Directory found and terminates

But It runs fine in windows

pradeeban, 2023-02-18:
Thanks, @parteekcoder, for your testing efforts.

./clear is required only for Docker executions (makedocker instead of makestudy). Otherwise, it can be skipped because we are going to run ./destroy to remove the whole directory.

https://control-core.readthedocs.io/en/latest/usage.html

Once we have the concore-editor integration, users will use the front-end to access these features. Then they won't have to deal with these scripts (unless they want to).

parteekcoder, 2023-02-18:
> Thanks, @parteekcoder, for your testing efforts.
>
> ./clear is required only for Docker executions (makedocker instead of makestudy). Otherwise, it can be skipped because we are going to run ./destroy [...]

yeah
Clear will clear the result only
And destroy will destroy the whole directory

As concore editor Also has this feature to clear
[image]

parteekcoder, 2023-02-18:
> Thanks, @parteekcoder, for your testing efforts.
>
> ./clear is required only for Docker executions (makedocker instead of makestudy). Otherwise, it can be skipped because we are going to run ./destroy [...]

thanks , I am going through whole codebase to understand better that will help me to integrate concore with concore-editor easily

pradeeban, 2023-02-18:
Yes, your understanding is right.

In the case of Docker run, running clear is a must. (otherwise, the names of the instances are used, and therefore the re-run will fail). Otherwise, we don't really need to run clear.

pradeeban, 2023-02-20:
I tested this on MacOS. It works.

Pradeebans-MBP:concore pradeeban$ ./makestudy demo/sample.graphml
python3 mkconcore.py demo/sample.graphml demo sample macos
mkconcore 22-09-18
concore path:      .
graphml input:     demo/sample.graphml
source directory:  demo
output directory:  sample
control core type: posix
treat .m as octave:False
MCR path:          /opt/matlab/MATLAB_Runtime_R2021a_Update_5/v910
Docker repository: markgarnold
Pradeebans-MBP:concore pradeeban$ cd sample
Pradeebans-MBP:sample pradeeban$ ls
build	clear	debug	maxtime	params	run	src	stop	unlock
Pradeebans-MBP:sample pradeeban$ tree
.
├── build
├── clear
├── debug
├── maxtime
├── params
├── run
├── src
│   ├── concore.hpp
│   ├── concore.py
│   ├── concore.v
│   ├── concore_default_maxtime.m
│   ├── concore_initval.m
│   ├── concore_iport.m
│   ├── concore_oport.m
│   ├── concore_read.m
│   ├── concore_unchanged.m
│   ├── concore_write.m
│   ├── controller.iport
│   ├── controller.oport
│   ├── controller.py
│   ├── import_concore.m
│   ├── mkcompile
│   ├── pm.iport
│   ├── pm.oport
│   └── pm.py
├── stop
└── unlock

1 directory, 26 files
Pradeebans-MBP:sample pradeeban$ ./build
Pradeebans-MBP:sample pradeeban$ tree
.
├── CU
├── CZ
│   ├── concore.iport
│   ├── concore.oport
│   ├── concore.py
│   ├── controller.py
│   ├── in1 -> ../PYM
│   └── out1 -> ../CU
├── PYM
├── PZ
│   ├── concore.iport
│   ├── concore.oport
│   ├── concore.py
│   ├── in1 -> ../CU
│   ├── out1 -> ../PYM
│   └── pm.py
├── build
├── clear
├── debug
├── maxtime
├── params
├── run
├── src
│   ├── concore.hpp
│   ├── concore.py
│   ├── concore.v
│   ├── concore_default_maxtime.m
│   ├── concore_initval.m
│   ├── concore_iport.m
│   ├── concore_oport.m
│   ├── concore_read.m
│   ├── concore_unchanged.m
│   ├── concore_write.m
│   ├── controller.iport
│   ├── controller.oport
│   ├── controller.py
│   ├── import_concore.m
│   ├── mkcompile
│   ├── pm.iport
│   ├── pm.oport
│   └── pm.py
├── stop
└── unlock

9 directories, 34 files
Pradeebans-MBP:sample pradeeban$ ./run
Pradeebans-MBP:sample pradeeban$ tree
.
├── CU
│   └── u
├── CZ
│   ├── __pycache__
│   │   └── concore.cpython-37.pyc
│   ├── concore.iport
│   ├── concore.oport
│   ├── concore.py
│   ├── concoreout.txt
│   ├── concorepid
│   ├── controller.py
│   ├── in1 -> ../PYM
│   └── out1 -> ../CU
├── PYM
│   └── ym
├── PZ
│   ├── __pycache__
│   │   └── concore.cpython-37.pyc
│   ├── concore.iport
│   ├── concore.oport
│   ├── concore.py
│   ├── concoreout.txt
│   ├── concorepid
│   ├── in1 -> ../CU
│   ├── out1 -> ../PYM
│   └── pm.py
├── build
├── clear
├── debug
├── maxtime
├── params
├── run
├── src
│   ├── concore.hpp
│   ├── concore.py
│   ├── concore.v
│   ├── concore_default_maxtime.m
│   ├── concore_initval.m
│   ├── concore_iport.m
│   ├── concore_oport.m
│   ├── concore_read.m
│   ├── concore_unchanged.m
│   ├── concore_write.m
│   ├── controller.iport
│   ├── controller.oport
│   ├── controller.py
│   ├── import_concore.m
│   ├── mkcompile
│   ├── pm.iport
│   ├── pm.oport
│   └── pm.py
├── stop
└── unlock

11 directories, 42 files
Pradeebans-MBP:sample pradeeban$ ./clear
Pradeebans-MBP:sample pradeeban$ tree
.
├── CU
├── CZ
│   ├── __pycache__
│   │   └── concore.cpython-37.pyc
│   ├── concore.iport
│   ├── concore.oport
│   ├── concore.py
│   ├── concoreout.txt
│   ├── concorepid
│   ├── controller.py
│   ├── in1 -> ../PYM
│   └── out1 -> ../CU
├── PYM
├── PZ
│   ├── __pycache__
│   │   └── concore.cpython-37.pyc
│   ├── concore.iport
│   ├── concore.oport
│   ├── concore.py
│   ├── concoreout.txt
│   ├── concorepid
│   ├── in1 -> ../CU
│   ├── out1 -> ../PYM
│   └── pm.py
├── build
├── clear
├── debug
├── maxtime
├── params
├── run
├── src
│   ├── concore.hpp
│   ├── concore.py
│   ├── concore.v
│   ├── concore_default_maxtime.m
│   ├── concore_initval.m
│   ├── concore_iport.m
│   ├── concore_oport.m
│   ├── concore_read.m
│   ├── concore_unchanged.m
│   ├── concore_write.m
│   ├── controller.iport
│   ├── controller.oport
│   ├── controller.py
│   ├── import_concore.m
│   ├── mkcompile
│   ├── pm.iport
│   ├── pm.oport
│   └── pm.py
├── stop
└── unlock

11 directories, 40 files
Pradeebans-MBP:sample pradeeban$ ./clear
rm: CU/*: No such file or directory
rm: PYM/*: No such file or directory
Pradeebans-MBP:sample pradeeban$

Please note the first execution of clear removed the CU/* and PYM/* successfully. The second execution throws the message it cannot execute since those files are not present (because the first execution is successful.)

I guess you were expecting to see more files to be deleted. But that is not the intended feature of clear. Clear removes the temp files CU/* and PYM/* as above in the local execution.

closed 2023-02-20 by pradeeban

parteekcoder, 2023-02-20:
I don't ran it directly rather I started the server and tested API using postman on MAC OS there this problem arises to me

pradeeban, 2023-02-20:
Ok, we use as above and not with Postman.