ControlCore-Project/concore #35: Output result gets overridden

parteekcoder, 2023-02-18:
Reproduce

1) run a study

Expected:
All Data generated , Should be present in the output files 

Actual:
Only last entry of result is present in the result files

pradeeban, 2023-02-18:
Thanks for the report, @parteekcoder. But this is the expected behavior for the concore controller implementation. Every result updates the previous, and the last one is presented.

If you want to keep track of the progress, that should be a separate implementation.

closed 2023-02-18 by pradeeban

parteekcoder, 2023-02-18:
> Thanks for the report, @parteekcoder. But this is the expected behavior for the concore controller implementation. Every result updates the previous, and the last one is presented.
> 
> If you want to keep track of the progress, that should be a separate implementation.

ok I thought that result files are to store all output generated , no problem thanks for reviewing