ControlCore-Project/concore #38: docker run failed

parteekcoder, 2023-02-19:
@pradeeban 

Reproduce

1. go to fri folder , build using docker command is given in DOCKER_README.md
2. Run the container

I get this:-  

[2023-02-19 16:00:46 +0000] [1] [INFO] Starting gunicorn 20.1.0
[2023-02-19 16:00:46 +0000] [1] [INFO] Listening at: http://0.0.0.0:8080 (1)
[2023-02-19 16:00:46 +0000] [1] [INFO] Using worker: sync
[2023-02-19 16:00:46 +0000] [22] [INFO] Booting worker with pid: 22
[2023-02-19 16:00:46 +0000] [24] [INFO] Booting worker with pid: 24
[2023-02-19 16:00:46 +0000] [22] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 589, in spawn_worker      
    worker.init_process()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/workers/base.py", line 134, in init_process 
    self.load_wsgi()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/workers/base.py", line 146, in load_wsgi    
    self.wsgi = self.app.wsgi()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 58, in load
    return self.load_wsgiapp()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 48, in load_wsgiapp   
    return util.import_app(self.app_uri)
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/util.py", line 359, in import_app
    mod = importlib.import_module(module)
  File "/opt/app-root/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/fri/main.py", line 8, in <module>
    from flask_cors import CORS, cross_origin
ModuleNotFoundError: No module named 'flask_cors'
[2023-02-19 16:00:46 +0000] [22] [INFO] Worker exiting (pid: 22)
[2023-02-19 16:00:46 +0000] [25] [INFO] Booting worker with pid: 25
[2023-02-19 16:00:46 +0000] [27] [INFO] Booting worker with pid: 27
[2023-02-19 16:00:46 +0000] [24] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 589, in spawn_worker      
    worker.init_process()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/workers/base.py", line 134, in init_process 
    self.load_wsgi()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/workers/base.py", line 146, in load_wsgi    
    self.wsgi = self.app.wsgi()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 58, in load
    return self.load_wsgiapp()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 48, in load_wsgiapp   
    return util.import_app(self.app_uri)
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/util.py", line 359, in import_app
    mod = importlib.import_module(module)
  File "/opt/app-root/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/fri/main.py", line 8, in <module>
    from flask_cors import CORS, cross_origin
ModuleNotFoundError: No module named 'flask_cors'
[2023-02-19 16:00:46 +0000] [24] [INFO] Worker exiting (pid: 24)
Traceback (most recent call last):
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 209, in run
    self.sleep()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 357, in sleep
    ready = select.select([self.PIPE[0]], [], [], 1.0)
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 242, in handle_chld       
    self.reap_workers()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 525, in reap_workers      
    raise HaltServer(reason, self.WORKER_BOOT_ERROR)
gunicorn.errors.HaltServer: <HaltServer 'Worker failed to boot.' 3>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/app-root/bin/gunicorn", line 11, in <module>
    sys.exit(run())
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 67, in run
    WSGIApplication("%(prog)s [OPTIONS] [APP_MODULE]").run()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/base.py", line 231, in run
    super().run()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/app/base.py", line 72, in run
    Arbiter(self).run()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 229, in run
    self.halt(reason=inst.reason, exit_status=inst.exit_status)
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 342, in halt
    self.stop()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 393, in stop
    time.sleep(0.1)
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 242, in handle_chld       
    self.reap_workers()
  File "/opt/app-root/lib/python3.6/site-packages/gunicorn/arbiter.py", line 525, in reap_workers      
    raise HaltServer(reason, self.WORKER_BOOT_ERROR)
gunicorn.errors.HaltServer: <HaltServer 'Worker failed to boot.' 3>

pradeeban, 2023-02-19:
This is probably due to the CORS implementation for the concore integration, @Rahuljagwani?

We did not really use Docker for FRI. So, there must be other bugs too! Perhaps, the solution is to just remove the Docker support for FRI altogether. Keeping the bug open.

parteekcoder, 2023-02-19:
> This is probably due to the CORS implementation for the concore integration, @Rahuljagwani?
> 
> We did not really use Docker for FRI. So, there must be other bugs too! Perhaps, the solution is to just remove the Docker support for FRI altogether. Keeping the bug open.

Actually there is one missing library , due to which container is not able run , I have added that

pradeeban, 2023-02-19:
Thanks.

One issue with your PRs is you keep adding changes to an open PR. That way, your fixes are bundling up. In this case, I would like to merge the PR instantly. But it is bundled up with a more demanding fix (fixing the execution for windows). Therefore, I cannot merge it.

Can you please make sure to create 1 PR for 1 issue, making sure no other fixes are in this PR?

Referring to #39:
This should just have that one line change, and nothing else in it. No fixes for other issues, no other enhancements, etc. To make the PR, you can create a feature branch from your main branch and make only that change, and create a PR from that branch.

Rahuljagwani, 2023-02-19:
> This is probably due to the CORS implementation for the concore integration, @Rahuljagwani?
> 
> We did not really use Docker for FRI. So, there must be other bugs too! Perhaps, the solution is to just remove the Docker support for FRI altogether. Keeping the bug open.

Actually in requirements.txt, the FLASK_CORS has to be included. It was already installed in my system so it worked for me. I will send a PR shortly after updating.

parteekcoder, 2023-02-19:
> Actually in requirements.txt, the FLASK_CORS has to be included. It was already installed in my system so it worked for me. I will send a PR shortly after updating.

You are right @Rahuljagwani but I have all ready created a PR for this issue

Rahuljagwani, 2023-02-19:
> You are right @Rahuljagwani but I have all ready created a PR for this issue

Ok, I missed that. You can proceed with your PR.

pradeeban, 2023-02-20:
Thanks for the fixes, both of you. Closing now.

closed 2023-02-20 by pradeeban