ControlCore-Project/concore #52: Ambiguity when two directories containing graphml file of same name are executed

Rahuljagwani, 2023-02-22:
@pradeeban 
As discussed in https://github.com/ControlCore-Project/concore/pull/50#issuecomment-1439137677, 'apikey' must be working and applicable for mentioned functions. 

**One of the test case:**
Suppose there are two directories named 'demo' and 'example' and both contains  'sample.graphml' . Now a user first executes  'demo/sample.graphml' a directory named 'sample' will be created and all functions (build, debug etc.) will be performed on it.
After this, if user wants to execute for 'example/sample.graphml', error will occur as two directories with same name cannot be formed inside a particular folder. Therefore, 'apikey' must be used to solve the same.

For now 'apikey' passed must be an optional argument as discussed.

Rahuljagwani, 2023-02-22:
I will be sending a PR regarding the same.

pradeeban, 2023-02-22:
Hi @Rahuljagwani,
The correct fix for the issue you mentioned is this:
Case 1:
./makestudy demo/sample.graphml
A folder "sample" is created above by default.

Case 2. The destination as a folder name specified as an argument:
./makestudy example/sample.graphml sample-whatever
A folder "sample-whatever" is created.

Both cases 1 and 2 are handled in the concore (when you run directly). You can test it (and let me know if something is not clear).

But currently in concore-editor, only case 1 is addressed, since the folder-name input is not provided as an entry. One fix could be, getting the input, "Output folder name" from the user.

The API Key addresses a different issue.
Case 3: Assume concore is hosted in a cloud server (we don't do that now - not enough security implemented). 

You run case 1, assuming you can do that (since you never had a "sample" folder for you after all.)
I also run case 1, assuming I can do that (since I never had a "sample" folder for myself either.)

You run case 2, assuming you can do that (since you never had a "sample-whatever" folder for you after all.)
I also run case 2, assuming I can do that (since I never had a "sample-whatever" folder for myself either.)

But this is a problem of multitenancy (since, we are sharing the folders across different users now.) So, appending the API Key functions as a means for Multitenancy. It makes folders such as sample-whatever-RahulAPIKey, sample-whatever-PradeebanAPIKey, sample-PradeebanAPIKey, ...

Since we don't really do this cloud deployment (that will need more technical consideration), we have not paid enough attention to API Key.

Now, what you should be doing to address your problem is, addressing Case 2. That is a separate input. Not API Key.

pradeeban, 2023-02-22:
For your case 2, since the output folder name is already considered by the makestudy, you should be able to use that variable without modifying the mkconcore.py.

Isn't it?

pradeeban, 2023-02-22:
I believe, even for this PR, you will only need to change FRI.

Handling the API Keys is a separate issue (which is less critical than the one you are encountering right now since that is a problem for "multitenancy" / multi-user executions only).

pradeeban, 2023-02-22:
But, since this "Output Directory Name" comes from the concore-editor input, that will require (non-trivial) edits to concore-editor. (This task, and several similar ones, to make the concore editor more complete, are part of the GSoC Project Idea 1).

Rahuljagwani, 2023-02-23:
> For your case 2, since the output folder name is already considered by the makestudy, you should be able to use that variable without modifying the mkconcore.py.
> 
> Isn't it?

Yes. I will send a separate PR regarding case 2, just by mentioning 'apikey' .

Rahuljagwani, 2023-02-23:
> Hi @Rahuljagwani, The correct fix for the issue you mentioned is this: Case 1: ./makestudy demo/sample.graphml A folder "sample" is created above by default.
>
> Case 2. The destination as a folder [...]

Actually PR #53 handles case1 and case3. For case 2 I will be sending a different PR. Anyway some code has to be changed in PR #53 to handle all 3 cases.

For now PR #53 will handle multitenancy for case 1 and case 3, which is of no use for now since concore is not deployed to cloud as you mentioned.

I understand why changes to mkconcore.py are being resisted, because if some new issue arises users will not even be able to call from CLI. Therefore, don't merge PR #53 but please don't close it. I will be working on same and after proper testing only I will ask for its review. 
Thanks.

Rahuljagwani, 2023-02-23:
> For your case 2, since the output folder name is already considered by the makestudy, you should be able to use that variable without modifying the mkconcore.py.
> 
> Isn't it?

Yes, few changes have to be done like making an argument name 'outdir' API calls of test.py and main.py in build function only.
And in test.py all the calls (build,debug etc.) will contain outer directory mentioned by user.

pradeeban, 2023-02-23:
> Yes, few changes have to be done like making an argument name 'outdir' API calls of test.py and main.py in build function only. And in test.py all the calls (build,debug etc.) will contain outer [...]

My suggestion is to have the first PR address just the FRI in that case. (which will be a mapping of the output dir argument, which is already present in makestudy and make it an optional parameter that can be passable on FRI). 

Handling the API Key is a separate issue, a separate PR (which probably should wait, especially since that seems to need edits outside FRI.)

Since the output dir name is optional, concore editor should work as-is (currently it is a hard-coded value after all). But the subsequent implementation (in GSoC) should fix the concore editor to get this as an input parameter too.

pradeeban, 2023-02-23:
//I understand why changes to mkconcore.py are being resisted, because if some new issue arises users will not even be able to call from CLI. Therefore, don't merge PR https://github.com/ControlCore-Project/concore/pull/53 but please don't close it. I will be working on same and after proper testing only I will ask for its review.//

Sounds good!

pradeeban, 2023-02-23:
Merged #54. Thanks!

I close this issue, @Rahuljagwani. I think we do not have the ambiguity issue anymore on FRI. Feel free to reopen, if you think I missed something.

The API Key is a separate issue.

closed 2023-02-23 by pradeeban

Rahuljagwani, 2023-02-23:
> Merged #54. Thanks!
> 
> I close this issue, @Rahuljagwani. I think we do not have the ambiguity issue anymore on FRI. Feel free to reopen, if you think I missed something.
> 
> The API Key is a separate issue.

Sure