ControlCore-Project/concore #62: Jupyter Lab is not installed, still test.py in FRI showing {"message":"Successfuly opened Jupyter"}

shivangvijay, 2023-03-12:
To replicate this issue:- uninstalled Jupter Lab (pip3 uninstall jupyterlab)

Now run 

$ cd conore/fri

$ cd server

$ python3 main.py

and 

$ cd ..

$ python3 test.py

Expected Result:- Should throw an error.

Actual Result:- showing {"message":"Successfuly opened Jupyter"}

Rahuljagwani, 2023-03-12:
Hello @shivangvijay 
I am not able to replicate this issue. Its giving error when jupyter is uninstalled.
Below is the error log.

127.0.0.1 - - [12/Mar/2023 21:48:21] "POST /openJupyter HTTP/1.1" 308 -
[2023-03-12 21:48:21,501] ERROR in app: Exception on /openJupyter/ [POST]
Traceback (most recent call last):
  File "/home/rahuljagwani/.local/lib/python3.10/site-packages/flask/app.py", line 2528, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/rahuljagwani/.local/lib/python3.10/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/rahuljagwani/.local/lib/python3.10/site-packages/flask_cors/extension.py", line 165, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "/home/rahuljagwani/.local/lib/python3.10/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/rahuljagwani/.local/lib/python3.10/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/rahuljagwani/concore-main/fri/server/main.py", line 218, in openJupyter
    proc = subprocess.Popen(['jupyter', 'lab'], shell=False, stdout=subprocess.PIPE, cwd=concore_path)
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)

shivangvijay, 2023-03-12:
Hello @Rahuljagwani, Which OS are you using?

Rahuljagwani, 2023-03-12:
> Hello @Rahuljagwani, Which OS are you using?

Ubuntu

shivangvijay, 2023-03-12:
I am getting below prints:-
In Python3 main.py
[image]

In Python3 test.py
[image]

My OS is Ubuntu 18.

Rahuljagwani, 2023-03-12:
I guess this is relevant issue. Even though I am getting error, I believe it is related to something else as you can see I am getting success code 308 in first line of my log.
> 127.0.0.1 - - [12/Mar/2023 21:48:21] "POST /openJupyter HTTP/1.1" 308 -

pradeeban, 2023-03-15:
We have not tested the JupyterLab integration adequately. No wonder if it has some bugs.

shivangvijay, 2023-03-29:
I spend lot of time to resolve this issue but didn't able to crack perfect solution with suprocess. 
There is one way to raise uninstall error code through subprocess:-

```
     try:
        proc = subprocess.run("jupyter lab", shell=True, check=True, capture_output=True)
    except subprocess.CalledProcessError as e:
        print(e)
```

In documentation of subprocess.run, they mentioned "If check is True and the exit code was non-zero, it raises a
    CalledProcessError. The CalledProcessError object will have the return code
    in the returncode attribute, and output & stderr attributes if those streams
    were captured."

But there is one issue with this code when jupyter lab is install the code (subprocess.run) wait to finish the process but it not happen, so no response goes to test.py.

Same we can do with subprocess.popen using communicate function. I explored other subprocess.check_call etc... nothing helping me.

shivangvijay, 2023-03-29:
I found one more issue:- 
I commented all the functions other than jupyter notebook.

[image]

then run the main.py and test.py

In main.py terminal I am getting two times success code.

```
127.0.0.1 - - [29/Mar/2023 15:33:24] "POST /openJupyter HTTP/1.1" 308 -
127.0.0.1 - - [29/Mar/2023 15:33:24] "POST /openJupyter/ HTTP/1.1" 308 -
```

pradeeban, 2023-03-29:
Thanks, @shivangvijay, for your findings.

I am honestly wondering whether we should get rid of Jupyter. It is just there installed. No one used it, and not properly tested.

Perhaps, proper integration and testing of Jupyter for concore (not merely installing - but using it to construct and run concore programs effectively, somewhat replacing the concore-editor's program editor) can be a GSoC 2024 project.

pradeeban, 2023-03-29:
Created a separate issue to contemplate what to do with Jupyter in the long run - #66.

Deeppjp116, 2023-05-29:
Hello,
I'm Deep.

        I'm interested in working on this issue. Could you please explain the problem and guide me on setting up the repository on my computer? I'm eager to contribute and give my best effort to resolve this. This will be my first valuable contribution. Thank you for giving me this opportunity.

pradeeban, 2024-03-28:
@Harshal-Rembhotkar, we are not sure how to proceed with this issue (if you read the discussions above, you will find that as well).

So, it may not be the right time to work on this issue.

Harshal-Rembhotkar, 2024-03-29:
Okay

closed 2026-02-03 by pradeeban