ControlCore-Project/concore #63: In test.py, after giving input 1 (with space), run method initiate

shivangvijay, 2023-03-12:
To replicate the issue:- 

run the FRI as a server:
````
$ cd conore/fri
$ cd server
$ python3 main.py
````
and 
````
$ cd ..
$ python3 test.py

when this code ask for input:- "methods - 1 for debug, 0 for run :"

Give input:-  <space>1 
This input not run debug, it initiate run method only.

Expected result:- Should initiate debug method as input is given 1.

Actual result:- It initiate run method.

More information:- 
The problem is actually due to input treating as string, so space also matter. If user give input <space><space>1, test.py treat it as 
"  1", which is not equal to "1". so the code goes on else condition and run method initiate.

method = input("methods - 1 for debug, 0 for run :")
if method == "1":
  debug("sample-anyname", "xyz")
else:  
  run("sample-anyname", "xyz")

pradeeban, 2023-03-15:
Thanks for the catch, @shivangvijay. I see you have a fix. If you make a pull request, I will merge it right away.

pradeeban, 2023-03-15:
Fixed in #61,

closed 2023-03-15 by pradeeban