ControlCore-Project/concore #67: A docker-compose file for FRI with Kong

pradeeban, 2023-03-29:
"The Kong instructions in the Docker README here can be confusing."

To avoid this, we can develope the docker-compose file. Also, it helps users to build and run multiple container with one command.

_Originally posted by @shivangvijay in https://github.com/ControlCore-Project/concore/issues/44#issuecomment-1488362559_

shivangvijay, 2023-03-29:
```
version: '3'

services:
  fri:
    build:
      context: .
    image: fri
    ports:
      - 8090:8080
    command: tail -f /dev/null

  kong-database:
    image: cassandra:3
    ports:
      - 9042:9042

  kong:
    image: kong
    depends_on:
      - kong-database
    environment:
      KONG_DATABASE: cassandra
      KONG_PG_HOST: kong-database
      KONG_PG_PASSWORD: kong
      KONG_CASSANDRA_CONTACT_POINTS: kong-database
      KONG_PROXY_ACCESS_LOG: /dev/stdout
      KONG_ADMIN_ACCESS_LOG: /dev/stdout
      KONG_PROXY_ERROR_LOG: /dev/stderr
      KONG_ADMIN_ERROR_LOG: /dev/stderr
      KONG_ADMIN_LISTEN: 0.0.0.0:8001, 0.0.0.0:8444 ssl
    ports:
      - 80:8000
      - 8443:8443
      - 8001:8001
      - 8444:8444

  ```

@pradeeban  I write the docker-compose file. Don't have full knowledge of kong and the exact output. Still, I think, it can be be good starting point for other developers. So I have going to put PR for this .

pradeeban, 2023-03-29:
Merged. Thanks. 
It is a new file. So it cannot break anything and can be a good starting point for future developments indeed.