ControlCore-Project/concore #71: Referenced resp before assignment

parteekcoder, 2023-05-12:
In the main.py, at the build method `resp` is returned as a response. But it can be the case when it is not get assigned by any value because they are inside the if block

parteekcoder, 2023-05-12:
2023-05-12 05:22:20,426: Exception on /build/test [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/parteek/.local/lib/python3.10/site-packages/flask_cors/extension.py", line 165, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/parteek/concore/fri/server/main.py", line 101, in build
    return resp
UnboundLocalError: local variable 'resp' referenced before assignment

Rahuljagwani, 2023-05-12:
Hi @parteekcoder 
I found out that this error arises if some user executed a build on any particular graphml, and forgot to destroy it. When the same graphml file will be executed by user, this error will come as the directory_path created already exists https://github.com/ControlCore-Project/concore/blob/main/fri/server/main.py#LL80C37-L80C37,
what u think can be done to solve this? I mean what can be the generic response if an already existing directory is tried to build again?

parteekcoder, 2023-05-13:
@Rahuljagwani I think we send message something like 'directory already exists' or can discuss

saksham-gera, 2025-01-24:
@Rahuljagwani @parteekcoder @pradeeban I Think The Issue Is Already fixed, 
If You're Expecting anything else from this issue, please let me know, I'll Fix It

pradeeban, 2025-01-31:
Thanks, @saksham-gera. Yes, looks like we just forgot to close it.

closed 2025-01-31 by pradeeban