ControlCore-Project/concore #97: Conntribute Action is not working

pradeeban, 2023-08-02:
We have merged the dev branches of concore and concore-editor to the main. We also deployed the latest concore-editor in github. We pulled the latest main of concore and ran fri.

When we click Contribute button, opening a study (concore/demo/sample.graphml), filling the fields appropriately (as listed in the README), it fails.

In the front-end, the error message always is: "Some error occurred, please try again after some time"
The backend FRI logs are as below: 
127.0.0.1 - - [02/Aug/2023 15:55:25] "OPTIONS /contribute HTTP/1.1" 200 -
127.0.0.1 - - [02/Aug/2023 15:55:28] "POST /contribute HTTP/1.1" 501 -

This is now deployed in the main CONTROL-CORE project github.io. Can you please have a look?

@parteekcoder

parteekcoder, 2023-08-02:
yeah sure, I will examine this

parteekcoder, 2023-08-03:
> @parteekcoder <https://github.com/parteekcoder>, I have merged your
> changes. I also made the logs verbose as you suggested and committed the
> changes too! (so we won't have to add code later to test. These logs can
> stay). All code in main now.
>
> Still failing. The error displayed now is,
>
> An error encountered: 'utf-8' codec can't decode byte 0xf0 in position
> 1133: invalid continuation byte
>
> The operating system is Mac.
>
> I think you trying this from another computer where you have not signed
> into your GitHub accounts will be a good place to test from.

Ok I will try it in MAC . Did you entered the same parameters as you sent
me last night? what is the filepath you entred

pradeeban, 2023-08-03:
Yes, and it fails regardless of what I enter, with the same error.

parteekcoder, 2023-08-03:
Hello Pradeeban,

I have checked the issue on MacOS Ventura - Python 3.11.3 and 3.10 - on
both conda and pip interpreters. It works fine. Please let me know which
python version you are on to check it forward.

pradeeban, 2023-08-04:
The version is 3.7.4. People use concore with Python 3.7 onward. So we should support those as well. Please check.

parteekcoder, 2023-08-04:
Hi Pradeeban,

Now I tested this with Python versions 3.6, 3.7, and 3.11 it is working
fine. And also tested it on MAC OS it working correctly.

But I made some changes that I think could solve this problem and added
some error logs so that I get to know what is happening at your system.

I request you check it once before the meeting and let me know , so that I
can fix it before the meeting.
PR:  https://github.com/ControlCore-Project/concore/pull/99

Thanks for your Invaluable time

pradeeban, 2023-08-04:
Ok, the example folder is working. Then I created another foloder (basically renaming "example" folder as "bingbing" and renaming all the files. That also works.

But when I try a study in the "demo" folder, it fails with the message "token decoded successfully 'utf-8' codec can't decode byte 0xf0 in position 1133: invalid continuation byte Some error Occured.Please try again after some time."

Is it because there are way too many files in the demo folder? If so, please explicitly mention in the ReadTheDocs documentation that the user must limit the folder to have only related files and not too many files.

Pls continue to test the Contribute Action for the next week with various studies and folder structures (and not just example folder). Then, you will be able to find solutions to generalize this for more files too.

Let's discuss today.

I am closing this issue since Contribute action is working, although with certain assumptions that must be expressed in the document in ReadTheDocs.

closed 2023-08-04 by pradeeban

parteekcoder, 2023-08-04:
Thanks, Pradeeban for your valuable time in testing. Actually it is not
because of the folder size. I know the issue why it happened and will
create a PR to fix this.

pradeeban, 2023-08-04:
Awesome. Thank you!

PR #105 (2023-08-07): Fix contribute action for larger folders
Fixes #97 properly