ControlCore-Project/concore #114: Refactoring Python code - Improve Readability and maintainability

pratik-choudhari, 2024-03-28:
**Who is the issue affecting?**
--
Contributors and Maintainers

**What is affected by this issue?**
--
Readability, maintainability, debugging and code reviews

**Comments**
--
Hi @pradeeban, I was trying to read the code and figure out its workings. I had a hard time understanding the core Python modules.
Common issues noticed:
- type hinting
- function docstrings
- error handling
- complicated equality expressions
- string formatting

**Sample Solution**
--
The `concore.write` function can be refactored as follows (original on the left, modified in the right):
[image]

**Scope**
--
Throughout the project

pratik-choudhari, 2024-03-28:
Additionally, If docstring, type hints and return types are added to functions, IDEs such as PyCharm can use this information and warn a developer if the parameters do not match with the type hints

pradeeban, 2025-02-04:
While this is a valid issue of a concore repository, this is a core file in the project. Changing it will need approvals from multiple reviewers and we will need to make sure nothing is mistakenly broken during the commit. Closing as "Won't Fix" for now.

closed 2025-02-04 by pradeeban