ControlCore-Project/concore #117: warning is raised when running the Flask application on non-Windows platforms

saksham-gera, 2025-01-16:
SyntaxWarning: invalid escape sequence '\l'
  proc = subprocess.check_output(["..\library", library_path, filename], shell=True, cwd=dir_path)

Steps to Reproduce
	1.	Run the Flask application on macOS/Linux.
	2.	Observe the SyntaxWarning in the logs when the app initializes.

Expected Behavior
	•	No warnings are raised.
	•	Path handling is platform-agnostic and does not produce errors or warnings on non-Windows platforms.

Environment
	•	OS: macOS/Linux
	•	Python Version: 3.8+
	•	Flask Version: 2.x

saksham-gera, 2025-01-16:
Hi @pradeeban ,

I noticed this issue and would like to take it up if that’s okay with you. Let me know if I can proceed with resolving it!

Thanks! 😊

pradeeban, 2025-01-31:
@saksham-gera sure, go for it.

pradeeban, 2025-02-04:
Thanks for the fix.

PR #118 (2025-02-04): Fixed Warning While Running Flask Server On Non-Windows Platform, improved test.py
fix #117 , by converting the strings to raw strings wherever required
also test.py is improved now
 - i fixed download function test by correctly handling the response, however the download function still cant find the file that needs to be downloaded
 - removed deprecated unused import of CGI
closed 2025-02-04 by pradeeban