ControlCore-Project/concore #120: Change Default Port 5000 to Avoid Conflict with AirDrop and AirPlay on macOS

saksham-gera, 2025-02-03:
Currently, the project uses port 5000 as the default for running the application. However, on macOS, this port is already used by AirDrop and AirPlay, which can cause conflicts, making it difficult for users to run the application smoothly.

To ensure seamless functionality across all platforms, including macOS, we should update the default port to an alternative, such as port 5050 or 8000.

**Key Changes Required:**

1. Update the default port in all configuration files, scripts, and documentation.
2. Modify any environment variables or constants that reference port 5000.
3. Ensure compatibility with existing workflows by providing instructions on overriding the port if necessary.

**Steps to Reproduce Issue:**

1. cd concore/fri/server
2. python3 main.py
3. Check for conflicts with AirDrop/AirPlay.
4. The service may fail to start or exhibit unexpected behavior.

**Environment:**

1. OS: macOS.
2. Python Version: 3.8+
3. Flask Version: 2.x

saksham-gera, 2025-02-03:
@pradeeban Sir, I would be happy to fix this issue along with some other pending issues. Please let me know if this needs to be addressed or if any changes are required. Looking forward to your guidance. 😊

pradeeban, 2025-02-04:
@saksham-gera this port is referred to in several places (including concore-editor repository at https://github.com/ControlCore-Project/concore-editor), and used by several users. Such an update will break backward compatibility. None of our users really used Airdrop or Airplay, even though we were developing in Mac OS. If one must use Airplay, they could change its port instead. ;) https://superuser.com/questions/1710207/macos-monterey-how-to-change-port-on-which-airplay-receiver-is-running

closed 2025-02-04 by pradeeban