ControlCore-Project/concore #125: mkconcore.py Code Not Well Organized and Full of Redundancy

saksham-gera, 2025-02-08:
The current implementation of mkconcore.py has several areas that can be improved for better readability and maintainability. The script contains a significant amount of code duplication, lacks modularity, and has mixed concerns, making it hard to read and maintain.

**Current Status:** The refactored version of mkconcore.py is being tested on Windows, Ubuntu, and macOS to ensure it works correctly across different environments.

**Expected Benefits:**
- Readability: The code will be easier to read and understand.
- Maintainability: Modular code is easier to maintain and extend.

[image]

saksham-gera, 2025-02-08:
@pradeeban , please let me know if this needs to be done or if I should hold off on working on this issue for now.

pradeeban, 2025-02-18:
This will be a big testing responsibility. Avoid working on it for now.